iondbproject/iondb #101: TODO: This is a potential issue and needs to be tested on SAMD3

danaack, 2017-05-03:
https://github.com/iondbproject/iondb/blob/development/src/dictionary/dictionary.c#L197

Craftplorer, 2018-05-16:
I have an Arduino Due(SAM3X8E) and i have some wired behavior. My Arduino Uno(ATmega328P) seems to work just fine. I only can read the first entry i insert. All the others return err_item_not_found.

`
#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __ORDER_LITTLE_ENDIAN__ 1234
#include <IonDB.h>

void setup() 
{
  Serial.begin(9600);
  Serial.println("Hello World!");

  Dictionary < int, int > *dict = new SkipList < int, int > (-1, key_type_numeric_signed, sizeof(int), sizeof(int), 7);

  dict->insert(3, 10);
  dict->insert(4, 16);
  dict->insert(5, 20);

  int value = dict->get(3);

  if(dict->last_status.error != err_ok)
  {
    Serial.println("3Something went wrong!");
    if(dict->last_status.error == err_item_not_found)
    {
      Serial.println("\tItem was not found");
    }
  }

  Serial.print("3 -> ");
  Serial.println(value);

  int newValue = dict->get(4);

  if(dict->last_status.error != err_ok)
  {
    Serial.println("4Somethign went wrong!");
    if(dict->last_status.error == err_item_not_found)
    {
      Serial.println("\tItem was not found");
    }
  }

  Serial.print("4 -> ");
  Serial.println(newValue);

}

void loop() 
{

}`

Arduino Due:
`Hello World!
3 -> 10
4Somethign went wrong!
	Item was not found
4 -> 0
`

Arduino Uno:
`Hello World!
3 -> 10
4 -> 16
`

Im not sure whats the acutall problem is. Your second links seems to show some problem with endian.
Also i saw [here](https://forum.arduino.cc/index.php?topic=199604.0) has some problem on this topic. I tried to add some lines before the include of iondb like this `#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__` but it did not change anything. Any idea what causes this behavior?

j4l-Shvn, 2020-11-02:
I have the same issue on a Teensy 3.6 board. Except in the setup part it comes out fine but if get is issued in the loop it issues the error as shown above. Dictionary is global in this case.

margaretselzer, 2021-07-19:
I have very similar issues on Due. Tried to play around with byte order defines, but nothing changed. When using SkipList it is possible to insert records and the all records query returns all of them. However with get and equality query only the first record is returned, all else fails. Insert works, but judging from the return of the all record query, the added keys are always placed on the end of the list and I think in a properly functioning list they would be added ordered (between the biggest smaller and the smallest bigger existing key). Update and delete - as expected - only works on the first element in the list. Interestingly it is possible to delete the whole list if one keeps deleting the fist element (as returned by the all records query). Range query also only works for the first element, but only if the upper limit is exactly the same as the key of the first element. In all other cases the range query returns nothing.

When tried OpenAddressHash, even the initial inserts do not work. I get back as many records for the all records query that is defined as dictionary size, and all of the returned keys and values are random numbers reflecting uninitialized memory space. Something seems to be fundamentally wrong with how the memory is managed.

None of the file based versions worked on any level. It may be due to the fact that I tried to use the SdFat library instead of the "default" SD library, but as I could never verify correct functioning with memory based versions, it is a moot point.

It feels like the solution is something trivial - e.g. a define at the right place could bring the memory management logic back to working. Unfortunately after 4 days of trying I need to abandon my attempt to make it work as writing my own db at this point feels a safer bet than investing more time in this, especially that it seems nothing really happened in the past 3 years with this project.

It's a shame though... it felt so close. :(

ArduinoMKR, 2022-04-02:
Hi,
I also had troubles even with the provided BasicUsage example on Arduino MKR series (SAMD21), with Arduino UNO everything works fine. Finally I got the solution for me, at least with the SkipList.

First I found, that the keys in the list should be stored in an ascending order by the "sl_insert" routine regardless in which order they are inserted. With the UNO it works, but with the MKR they are just stored in order they were inserted. You can have a look on all values you have stored with a defined Cursor and then using "_dict-->allRecords()_". They have to be shown in an ascending order.
So I found, that there is something wrong with the insert routine and I had a closer look on it. The thing is simply the following:
The decision where to insert a KV with the so called "vanilla insert" (= "insert before") doesn't work on SAMD, because the return value of the compare function is defined as an an 8 bit CHAR. But it is intended to return the compare results  SIGNED (-1/smaller, 0/equal, 1/greater) but the (-1/smaller) is returned as "255" and the comparison for example in skip_list.c in line 191 leads to a misinterpretation because 255 is greater than 0. On AVR/UNO this mechanism works.

You can see this in sl_insert on Line 192:
_while (NULL != cursor->next[h] && skiplist->super.compare(key, cursor->next[h]->key, key_size) **>= 0**)_ 
--> leads to "true" where it shoudn't

Same effect on line 326 in the sl_find_node routine.

You could be done with a simple typecast on this two lines if you are only interested in the skipList
**(int8_t)** _skiplist->super.compare(key, cursor->next[h]->key, key_size) >= 0)_

I did it in a more central way, so I changed the return types of the compare functions from char to int8_t in:

- dictionary_types, line 83: typedef **int8_t** (*ion_dictionary_compare_t)(
- dictionary.h. lines 219 and 246 for the declaration of dictionary_compare_unsigned_value and dictionary_compare_signed_value
- dictionary.c, lines 206 and 238 for the above mentioned routines

Surely there may be better solutions but for me in practice it works.

Hope I could help a little bit
Giorgio

ArduinoMKR, 2022-05-22:
margaretselzer wrote:
> It feels like the solution is something trivial - e.g. .......

Your feeling is right :-)
I got the FileBasedUsage example running on an Arduino MKR with MKR Mem Shield, with my modification mentioned above and with a simple modification as follows:

In the file _ionDB/src/file/sd_stdio_c_iface.cpp_ You need to change _line 153_
from: _operation = FILE_WRITE;_
to:     _operation = (O_RDWR  | O_CREAT);_
That's all.

Why? 
The reason is, that the FILE_WRITE definition in the "default" SD library includes _O_APPEND_ as well, but in that mode the fseek mechanism isn't applicable for writing to an arbitrary position, so leave it away. 
See also e.g.: https://stackoverflow.com/questions/10631862/fseek-does-not-work-when-file-is-opened-in-a-append-mode

margaretselzer, 2022-05-24:
Unfortunately this is not it. Probably my fault that I was not clear enough in my original post, but the symptoms that I describe has nothing to do with files. They were all the memory based versions. I explicitly write that the file based versions did not work on any level. Well, in case of files your solution might be a step forward, but if the memory based version is not working then there are some other issues at play.

ArduinoMKR, 2022-05-25:
> They were all the memory based versions...

Yes, I got your point. And because of this I wrote in my last post from 22. May "with my modification **above**". _Above_ should mean a relation to my post from 2 Apr. where I described how I got the _SkipList_ working properly with _Equality_ and _Range_ on Arduino MKR with the _BasicUsage_ example. It's about a problem with variable types which needs to be corrected. Sorry for my unclearity in my last post.