iondbproject/iondb #131: Require help instructions for build with espressif sdk

usermicrodevices, 2017-10-16:
Hello,
i copy IonDB to my esp32 arduino directory "components\arduino\libraries\IonDB",
add include line #include "IonDB.h" or #include <IonDB.h> before #include "Arduino.h" or after,
but get error "main/example.h:12:19: fatal error: IonDB.h: No such file or directory",
if write full path, then not found other IonDB source files,
where i can get help for build with ESP32 FreeRTOS SDK?

danaack, 2017-10-17:
Hi @usermicrodevices! 
Did you clone the IonDB repository or did you install the IonDB library from the IonDB Library Manager?

usermicrodevices, 2017-10-18:
Hi @danaack!
I copy IonDB to my esp32 SDK project directory. This is not Arduino IDE project, it is msys toolchain with esp32 SDK, any from https://github.com/espressif/esp-idf/tree/master/examples is same, i only append https://github.com/espressif/arduino-esp32 to "components" subdirectory.
Now i put IonDB sources to near main cpp file, and get long list errors.
[main.zip](https://github.com/iondbproject/iondb/files/1393948/main.zip)