iondbproject/iondb #136: compilation error

tkona, 2018-03-01:
Hi,

I tired you library, but I am getting fatal error:

danaack, 2018-03-01:
Hi @tkona,

Thanks for reaching out! What was your method of cloning the library, are you using v2.20 from https://www.arduinolibraries.info?

tkona, 2018-03-01:
Hi, Danaack,

no, I dowloaded it from github. Found it on google..

Shell I use the one you provided a link for?

Best regards,
Natko

danaack, 2018-03-01:
If you will be compiling on Arduino, that will be the simplest option for you. You can accomplish this by either:

- Downloading the .zip file from https://www.arduinolibraries.info/libraries/ion-db

- Or in the Arduino IDE selecting Sketch -> Include Library -> Manage Libraries... -> search for IonDB

Example use sketches can be found in the examples folder or File -> Examples -> IonDB.

If you are not compiling on Arduino, please let me know what error you were getting and I can try and help you further.

tkona, 2018-03-01:
Thank you very much for helping.
It is compiled for esp32.
I will try this tomorrow, and let you know.

Best regards,
Natko

tkona, 2018-03-02:
>    info/libraries/ion-db

Hi, Danaack,

I tried downloading as suggested, tried 2.2.0 and 2.1.3, but getting the
same errors:
(compiled for ESP32)

sd_stdio_c_iface.cpp: In function SD_FILE* sd_fopen(char*, char*)

sd_stdio_c_iface.cpp: 124:13: error: invalid conversion from 'const char*'
to 'uint8_t {aka unsigned char}' [-fpermissive]
   operation = FILE_READ

sd_stdio_c_iface.cpp: 136:13: error: invalid conversion from 'const char*'
to 'uint8_t {aka unsigned char}' [-fpermissive]
   operation = FILE_WRITE

sd_stdio_c_iface.cpp: 144:13: error: invalid conversion from 'const char*'
to 'uint8_t {aka unsigned char}' [-fpermissive]
   operation = FILE_WRITE

sd_stdio_c_iface.cpp: 153:13: error: invalid conversion from 'const char*'
to 'uint8_t {aka unsigned char}' [-fpermissive]
   operation = FILE_WRITE

sd_stdio_c_iface.cpp: 156:13: error: invalid conversion from 'const char*'
to 'uint8_t {aka unsigned char}' [-fpermissive]
   operation = FILE_WRITE

sd_stdio_c_iface.cpp: 164:41: error: invalid conversion from 'uint8_t {aka
unsigned char}' to 'const char*' [-fpermissive]
   (file)->f = SD.open(filename, operation)

SD.h:17: In file included from
sd_stdio_c_iface.cpp:41: from
FS.h:88: note    initializing argument 2 of fs  File fs  FS  open(const
char*, const char*)
   File open(const char* path, const char* mode = FILE_READ)

sd_stdio_c_iface.cpp: In function size_t sd_fread(void*, size_t, size_t,
SD_FILE*)

sd_stdio_c_iface.cpp: 185:59: error: invalid conversion from 'char*' to
'uint8_t* {aka unsigned char*}' [-fpermissive]
   int num_bytes = stream->f.read((char *) ptr, size * nmemb)

SD.h:17: In file included from
sd_stdio_c_iface.cpp:41: from
FS.h:58: note    initializing argument 1 of size_t fs  File  read(uint8_t*,
size_t)
   size_t read(uint8_t* buf, size_t size)
Error compiling libraries

Best regards,
Natko

tkona, 2018-03-02:
Hi, Danaack,

I have change the library, so it compiles now.
However, when using it as a Ditctionary<char, int>, I am getting wrong
results from the get function.

For example,

Dictionary < char, int > *m_Numbers = new SkipList<char, int>(-1,
key_type_char_array, sizeof(char), sizeof(int), 21);

m_Numbers->insert('0', 0);
m_Numbers->insert('1', 1);
m_Numbers->insert('2', 2);
m_Numbers->insert('3', 3);
m_Numbers->insert('4', 4);
m_Numbers->insert('5', 5);
m_Numbers->insert('6', 6);
m_Numbers->insert('7', 7);
m_Numbers->insert('8', 8);
m_Numbers->insert('9', 9);
m_Numbers->insert('A', 10);
m_Numbers->insert('C', 11);
m_Numbers->insert('E', 12);
m_Numbers->insert('F', 13);
m_Numbers->insert('H', 14);
m_Numbers->insert('J', 15);
m_Numbers->insert('L', 16);
m_Numbers->insert('P', 17);
m_Numbers->insert('S', 18);
m_Numbers->insert('U', 19);
m_Numbers->insert('-', 20);

m_Numbers->get('A') - returns 4, should be 10,
m_Numbers->get('C') - returns 10, should be 11,
m_Numbers->get('-') - returns 10, should be 20...

Best regards,
Natko

danaack, 2018-03-02:
Hi Natko,

I'm glad you were able to get it compiling! As for the wrong value being returned, I tried it out on Arduino and I was able to get the correct output with the following sketch:

```#include <IonDB.h>

void
setup(
) {
	Serial.begin(9600);

  Dictionary < char, int > *m_Numbers = new SkipList<char, int>(-1,
  key_type_char_array, sizeof(char), sizeof(int), 21);

  m_Numbers->insert('0', 0);
  m_Numbers->insert('1', 1);
  m_Numbers->insert('2', 2);
  m_Numbers->insert('3', 3);
  m_Numbers->insert('4', 4);
  m_Numbers->insert('5', 5);
  m_Numbers->insert('6', 6);
  m_Numbers->insert('7', 7);
  m_Numbers->insert('8', 8);
  m_Numbers->insert('9', 9);
  m_Numbers->insert('A', 10);
  m_Numbers->insert('C', 11);
  m_Numbers->insert('E', 12);
  m_Numbers->insert('F', 13);
  m_Numbers->insert('H', 14);
  m_Numbers->insert('J', 15);
  m_Numbers->insert('L', 16);
  m_Numbers->insert('P', 17);
  m_Numbers->insert('S', 18);
  m_Numbers->insert('U', 19);
  m_Numbers->insert('-', 20);

	/* Retrieve a value by providing a key */
	int my_value1 = m_Numbers->get('A');

	/* You should check the status on every operation to ensure good data integrity */
	if (err_ok != m_Numbers->last_status.error) {
		printf("Oh no! Something went wrong with my get operation\n");
	}

	printf("A -> %d\n", my_value1);

  my_value1 = m_Numbers->get('C');

  /* You should check the status on every operation to ensure good data integrity */
  if (err_ok != m_Numbers->last_status.error) {
    printf("Oh no! Something went wrong with my get operation\n");
  }

  printf("C -> %d\n", my_value1);

  my_value1 = m_Numbers->get('-');

  /* You should check the status on every operation to ensure good data integrity */
  if (err_ok != m_Numbers->last_status.error) {
    printf("Oh no! Something went wrong with my get operation\n");
  }

  printf("- -> %d\n", my_value1);
	
	/* Always clean up your dictionaries when you're done with them! */
	delete m_Numbers;
}

void
loop(
) {}```

Was your code similar?

tkona, 2018-03-03:
Hi Danaack,

Yes, almost the same, only without error checking after the get method.
However, the result of the get method is totally different.

I don't have time right now to debug it thoroughly, and I ended writing my
own methods, but I will try to help identify the problem when I catch some
spare time.

Thank you for your cooperation.

Best regards,
Natko