iondbproject/iondb #143: Examples fail to compile on ESP-8266

inetBrad, 2019-12-14:
The examples fail to compile.

SimpleUsage - Generates an error on every insert() using Arduino IDE 1.8.10 and board set to NodeMCU 1.0 (ESP-12E Module). No error when setting the board to Arduino.

```
StringUsage:15:23: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
  dict->insert(3, "one");
```

BahramSoheili, 2020-01-05:
you need to cast each string value in C++ by (ion_value_t) 
for example:
    dict->insert(3, (ion_value_t) "one");
    dict->insert(4, (ion_value_t) "two");
    dict->insert(9,  (ion_value_t) "six");
    dict->insert(42,  (ion_value_t) "ten");

BahramSoheili, 2020-01-05:
you need to cast each string value in C++ by (ion_value_t)
for example:
dict->insert(3, (ion_value_t) "one");